metriful/sensor #1: AQI accuracy & P2 from PPD42

Sennevds, 2020-09-17:
Hi,

Really love the board and I got everything up and running and sending data via MQTT to my Home Assistant. Only have 2 questions:

1. I can't get the AQI accuracy above 1 it's most of the time 1 and sometimes even 0. I've already blow on the sensor but it looks like nothing works
2. The PPD42 has a port 2 if you like at this https://github.com/mauricecyril/pidustsensor it's used for reading PM1.0 values. Is there an option to use this port?

metriful, 2020-09-17:
Hello!

1. It can take a long time, especially for new sensors (there can be a "burn-in" time for stabilization). It is important to keep the sensor running frequently (use 3 second cycles). Expose to clean air and dirty air, for example alcohol vapor.

2. The MS430 board has a single input for particle sensing. You can connect it to the PPD42 pin 2 instead of pin 4, both pins use the same output signal method. However, the manufacturers do not give much information about pin 2 or the related threshold input on pin 5.

Sennevds, 2020-09-17:
Thanks for the answers. The accuracy went after 5 hours to 3 and then after a few hours back to 0. Is this normal?  I have the refresh on 100sec at this moment. 
For port 2 I understand. Going to see if I can find more info on this.

metriful, 2020-09-19:
When the accuracy went from 3 to 0, did the MS430 board exit cycle mode, or could it have been reset? If either of these happens, the calibration starts again from accuracy 0.

metriful, 2020-09-19:
To summarize the accuracy value:

It can take several hours for a new sensor to stabilize and go to higher accuracy. The stabilization will be faster if you use a shorter cycle time. As the sensor ages it should be quicker.

If you exit cycle mode, or reset the board, it will start again from 0 and needs to recalibrate again. Otherwise the accuracy should rarely return to 0 after going to 1. 

If it remains 1 for a long time, expose the sensor to a marker pen, alcohol etc. The AQI should go to about 500 and then the accuracy will go to 2.

The accuracy will then slowly alternate between 2 and 3, this is intended.

Sennevds, 2020-09-19:
For me it variates between 2 3 and 1. It's running constantly with a cycle of 100sec. You can find my code I use here https://github.com/Sennevds/sensor/tree/master/Home_Assistant I don't think I do anything special. When I'm back from my holiday I'm going to make a separate sensor for the accuracy so I can better track the values.

metriful, 2020-09-19:
OK, that sounds like normal operation.

closed 2020-09-19 by metriful