metriful/sensor #5: IoT_cloud_logging.py

nsk, 2020-10-03:
pi@Metriful-CH341SER:~/sensor/Raspberry_Pi $ sudo python IoT_cloud_logging.py 
Traceback (most recent call last):
  File "IoT_cloud_logging.py", line 17, in <module>
    import requests
ImportError: No module named requests

and how activate  SDS011 sensor on IoT file ?? 

thx

metriful, 2020-10-03:
Hi,

**1.** The Raspberry Pi examples require Python packages called smbus and requests. These are already installed in recent versions of Raspberry Pi OS. Which operating system version are you using? 

It looks like you need to install requests, by doing:
```
pip3 install requests
```
If that does not work, you may need to install pip:
```
sudo apt-get install python3-pip
```
**2.** sudo should not be necessary to run the examples, just do:
```
python3 IoT_cloud_logging.py
```
**3.** To use the SDS011, connect it with 3 wires as explained in the readme, then in the file IoT_cloud_logging.py change the line:
```
particleSensor = PARTICLE_SENSOR_OFF
```
to:
```
particleSensor = PARTICLE_SENSOR_SDS011
```

nsk, 2020-10-03:
I use via usb

Sent from ProtonMail Mobile

metriful, 2020-10-03:
The SDS011 cannot be used via USB with the MS430 sensor board, or with the code examples.

To use the SDS with the MS430, you must connect 3 pins on the SDS using wires. This is explained in the GitHub Readme.
The wires go to GND and 3.3V on the Raspberry Pi and to the PRT pin on the MS430.

The USB cable is for using the SDS011 directly with a computer. We do not give code examples for this.

nsk, 2020-10-03:
ok

Alexandre Wellhoff
evopyro@protonmail.com
+41.78.637.79.83

Sent with [ProtonMail](https://protonmail.com) Secure Email.

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

nsk, 2020-10-03:
"Traceback (most recent call last):
  File "IoT_cloud_logging.py", line 18, in <module>
    from sensor_functions import *
  File "/home/pi/sensor/Raspberry_Pi/sensor_functions.py", line 15, in <module>
    import RPi.GPIO as GPIO
ModuleNotFoundError: No module named 'RPi'"

config file
#  IoT_cloud_logging.py
   
#  Example IoT data logging code for the Metriful MS430. 
#  This example is designed to run with Python 3 on a Raspberry Pi.
   
#  Environmental data values are measured and logged to an internet 
#  cloud account every 100 seconds. The example gives the choice of 
#  using either the Tago.io or Thingspeak.com cloud - both of these 
#  offer a free account for low data rates. 

#  Copyright 2020 Metriful Ltd. 
#  Licensed under the MIT License - for further details see LICENSE.txt

#  For code examples, datasheet and user guide, visit 
#  https://github.com/metriful/sensor

import requests
from sensor_functions import *

#########################################################
# USER-EDITABLE SETTINGS

# How often to read and log data (every 3, 100, 300 seconds)
# Note: due to data rate limits on free cloud services, this should 
# be set to 100 or 300 seconds, not 3 seconds.
cycle_period = CYCLE_PERIOD_100_S

# Which particle sensor, if any, is attached (PPD42, SDS011, or OFF)
particleSensor = PARTICLE_SENSOR_SDS011

# IoT cloud settings.
# This example uses the free IoT cloud hosting services provided 
# by Tago.io or Thingspeak.com
# Other free cloud providers are available.
# An account must have been set up with the relevant cloud provider and 
# an internet connection to the Pi must exist. See the accompanying 
# readme and User Guide for more information.

# Choose which provider to use
use_Tago_cloud = True 
# To use the ThingSpeak cloud, set: use_Tago_cloud=False

# The chosen account's key/token must be inserted below.  
if (use_Tago_cloud):
  # settings for Tago.io cloud
  TAGO_DEVICE_TOKEN_STRING = "30ffb9d4-98aa-419b-b33b-c856fdbd85d7"
else:
  # settings for ThingSpeak.com cloud
  THINGSPEAK_API_KEY_STRING = "PASTE YOUR API KEY HERE WITHIN QUOTES"

# END OF USER-EDITABLE SETTINGS
#########################################################

# Set up the GPIO and I2C communications bus
(GPIO, I2C_bus) = SensorHardwareSetup()

# Apply the chosen settings to the MS430
if (particleSensor != PARTICLE_SENSOR_OFF):
  I2C_bus.write_i2c_block_data(i2c_7bit_address, PARTICLE_SENSOR_SELECT_REG, [particleSensor])
I2C_bus.write_i2c_block_data(i2c_7bit_address, CYCLE_TIME_PERIOD_REG, [cycle_period])

#########################################################
[... 77 lines cut ...]
      payload[5] = {"variable":"bvoc","value":"{:.2f}".format(air_quality_data['bVOC'])}
      payload[6] = {"variable":"spl","value":"{:.1f}".format(sound_data['SPL_dBA'])}
      payload[7] = {"variable":"peak_amp","value":"{:.2f}".format(sound_data['peak_amp_mPa'])}
      payload[8] = {"variable":"illuminance","value":"{:.2f}".format(light_data['illum_lux'])}
      payload[9] = {"variable":"particulates","value":"{:.2f}".format(particle_data['concentration'])}
      requests.post(tago_url, json=payload, headers=tago_header, timeout=2)
    else:
      # Use ThingSpeak.com cloud
      payload = "api_key=" + THINGSPEAK_API_KEY_STRING 
      payload += "&field1=" + "{:.1f}".format(air_data['T_C'])
      payload += "&field2=" + str(air_data['P_Pa'])
      payload += "&field3=" + "{:.1f}".format(air_data['H_pc'])
      payload += "&field4=" + "{:.1f}".format(air_quality_data['AQI'])
      payload += "&field5=" + "{:.2f}".format(air_quality_data['bVOC'])
      payload += "&field6=" + "{:.1f}".format(sound_data['SPL_dBA'])
      payload += "&field7=" + "{:.2f}".format(light_data['illum_lux'])
      payload += "&field8=" + "{:.2f}".format(particle_data['concentration'])
      requests.post(thingspeak_url, data=payload, headers=thingspeak_header, timeout=2)
      
  except:
    # An error has occurred, likely due to a lost internet connection, 
    # and the post has failed.
    # The program will retry with the next data release and will succeed 
    # if the internet reconnects.
    print("HTTP POST failed.")

nsk, 2020-10-04:
root@Metriful-CH341SER:/home/pi/sensor/Raspberry_Pi# python3 IoT_cloud_logging.py
Traceback (most recent call last):
  File "IoT_cloud_logging.py", line 18, in <module>
    from sensor_functions import *
  File "/home/pi/sensor/Raspberry_Pi/sensor_functions.py", line 15, in <module>
    import RPi.GPIO as GPIO
ModuleNotFoundError: No module named 'RPi'
root@Metriful-CH341SER:/home/pi/sensor/Raspberry_Pi#

nsk, 2020-10-04:
resolve with install smbus2 via pip3

closed 2020-10-04 by nsk