metriful/sensor #10: Sending data over MQTT

marcob1281, 2020-10-19:
Hi guys 
Would it  be possible to get examples About sending data over MQTT?
I have my home iot server running Home Assistant Android I wanted to aggregate data of the sensors there.

Thanks

metriful, 2020-10-20:
Hi, we will consider adding this to the next code release, thanks.

closed 2020-10-22 by metriful

marcob1281, 2020-10-25:
I have found a quick solution based on NodeMCU ESP8266.
Starting from IoT cloud example, I have created a buffer for each value I want to transfer over MQTT.
Then each buffer content is sent to MQTT server using client.publish() function of PubSubClient library (Arduino).
It works quite well.

madcat-nl, 2020-12-20:
[mqtt_readout.py.txt](https://github.com/metriful/sensor/files/5721569/mqtt_readout.py.txt)
[sensor_functions.py.txt](https://github.com/metriful/sensor/files/5721571/sensor_functions.py.txt)

I also wanted to use MQTT, here are some modifications i did to the sensor_functions.
and created a mqtt_readout based on the cycle_readout.

Have fun

metriful, 2020-12-22:
Hi,

Very nice, thanks for sharing. MQTT is good but is maybe not suitable for all users, that's why we didn't include it.

Regards,
Customer support

Metriful Ltd.

admin@metriful.com
www.metriful.com

Metriful Ltd is registered in England & Wales. Company Registration Number: 12397348.
Registered office address: 27 Old Gloucester Street, London, WC1N 3AX, United Kingdom