metriful/sensor #11: Cycle mode intervals

holgerebert, 2020-10-23:
Hi,
is it possible to adjust the intervals? Looking for something between 3 and 100 seconds, 10 seconds and 30 seconds would be perfect for me.
Best regards,
Holger

metriful, 2020-10-23:
The cycle intervals of 3, 100, 300 seconds come from the Bosch air quality algorithm, so cannot be changed BUT you can achieve it in two other ways:

1) Use on-demand readout method in standby mode, and choose any interval (but then you do not get the air quality data)

2) Use a shorter cycle than the interval you want and update the data on every cycle. Separately run your output events at whatever interval you want, and use the most recent data. You can simply check the time while waiting for the data, or use a timer interrupt to trigger the output.

The basic method of reading, used in all the examples, is:

start of main loop
  wait for data
  read data
  output data
go to start of main loop

So there is one output on every cycle.

An example of method 2 to output the data every 10 seconds:
(Use a 3 second cycle)

start of main loop
  while (waiting for data) {
    if (time since last output >= 10 seconds) {
      output data
    }
  }
  read data
go to start of main loop

The output data can be up to 3 seconds old but this is usually not a problem.

holgerebert, 2020-10-23:
Ok, got it...😊
Thank you!

closed 2020-10-24 by holgerebert