metriful/sensor #13: CO2 value accuracy

marcob1281, 2020-10-29:
I may have some issue with CO2 value accuracy.
I am running the RPi code and monitoring CO2 value and accuracy value.
After all night long measuring in a closed empty room, I got a CO2 value in the nearby of 500ppm with AQI accuracy at 3.

This morning I have opened the window and CO2 value ramped up tu 1000+ ppm (I could expect a reverse behaviour ).....air here is quite clean, I am in the country side and measures done with other sensors in the past marked values around 410ppm.

Any suggestion?

metriful, 2020-10-30:
The CO2 value is an estimate, the sensor does not measure CO2 directly. The algorithm assumes that VOCs detected are due to people's breath and then uses a correlation between VOC and exhaled CO2. This is not always valid.

The air quality index is the main measurement because the sensor is directly measuring impurities in the air. The CO2 and breath VOC are estimates and only valid with certain assumptions - it is probably best to consider them as experimental only.

marcob1281, 2020-10-30:
ok got it.
thanks

Marco Barbato

closed 2020-10-30 by metriful