metriful/sensor #17: Air Quality Accuracy: self calibration ongoing.

ananchev, 2020-12-21:
I have been running for 3+ hours the sensor on esp8266 with the cycle readout example. Whilst there is value for Air Quality Index returned, still low accuracy is reported with self calibration ongoing. 
Is this normal and how long would the calibration process normally take?

ananchev, 2020-12-22:
This now goes for more than 15 hours. Read the instruction says it might take several minutes, but seems not the case for me. 
Could you please advise what could be causing this?

metriful, 2020-12-22:
This is normal and will improve as the sensor ages. To make it faster please try the following:

Use 3 second cycle. The stabilization will be faster if you use a shorter cycle time. As the sensor ages it should be quicker (impurities on the gas sensor are "baked out").

After running for about 30-60 minutes on 3-second cycle, if accuracy is still "low", expose the sensor to a solvent vapor (e.g. marker pen, alcohol etc). The AQI should go to about 500 and then the accuracy will probably go to "medium". If not, it needs longer running first.

The accuracy will slowly alternate between medium and high (and occasionally low). This is normal operation ("medium" is actually OK).

You can change to longer cycles after the accuracy has improved.

ananchev, 2020-12-22:
Thanks for your response! 
All looks good now - after it had run on 3s cycle last night the advice to expose to solvent vapour did the job. It indeed is alienating between high and medium (with self calibration).

closed 2020-12-22 by ananchev