metriful/sensor #23: Home assistant keeps 'kinda'crashing when using the ESP8266 script for HA

modular1, 2021-02-09:
I was wondering if this is a known problem, but as soon as my ESP8266 sends the data to HA seems to Crash i can only see the home screen which is not updated anymore and navigating to other sections results in a 'Error while loading page'. When i look in the console tab i see a 500 error. When i reboot HA(raspberry) it works untill it receives the next update from the sensor.
my config is:
Raspberry 3(2gig) and 4(4gig) (tried both)
HA 32 and 64 bit (tried both)

Any idea what cause this or or people reconize this?

metriful, 2021-02-10:
Hi. This problem has not been reported by anyone else.

Which version of Home Assistant are you using?

Please check that the token you put in the ESP8266 code is correct. Also try removing any symbols from the SENSOR_NAME parameter.

Do you have any automations that use the data? If yes, try disabling/deleting them.

Is the WiFi network created by the Raspberry Pi, or another device? Try using another network, e.g. a phone hotspot, to see if it is a network problem.

modular1, 2021-02-11:
Okay i moved my setup into kubernets and now problem seems solved. No idea wat the problem was. Sorry for the time

closed 2021-02-11 by modular1