metriful/sensor #24: Sensor stop sending data without apparent reason

rmazzali, 2021-05-06:
Hello,
I implemented a python program on a RPI Zero, that reads sensor values and send them over MQTT using Paho library. It's very similar to the one shared in another issue. It runs as a service.
It worked fine for weeks, then started to have problems: after a while (somtimes hours, others days or minutes) it stop sending data: the MQTT service is up and running, the python service is alive but data are not sent to MQTT broker.
I really cannot understand what happens, can you provide some advice to troubleshoot the sensor?

Best regards
Roberto

metriful, 2021-05-10:
Hi Roberto,

Do you get the same problem running one of our example programs (not as a service)?

You should add logging to your Python program to see what happens. For example, which part of the program is last reached. 

You should also check that all the wiring is secure with no loose connections. If you lose power on the Pi and/or sensor, does it restart correctly?

rmazzali, 2021-05-18:
Hello and thank you for your help. I think it was client.loop_stop() command in wrong place.
Here you can find my working code.

[MQTT_readout_PUBLIC.py.txt](https://github.com/metriful/sensor/files/6504181/MQTT_readout_PUBLIC.py.txt)

closed 2021-05-18 by rmazzali