metriful/sensor #26: Calibration of Air Quality Sensor

vincePeKaWay, 2021-07-10:
Hey, can you tell me how to calibrate the air quality sensor? It always tells me that calibration is incomplete. I'm running cycle_readout.py on a raspberry permanently btw.
Do I need to run another script to calibrate?

metriful, 2021-07-12:
Hi. The sensor needs to experience a range of clean and dirty air for the self calibration. To make it faster please try the following:

Use 3 second cycle. The stabilization will be faster if you use a shorter cycle time. As the sensor ages it should be quicker (impurities on the gas sensor are "baked out").

After running for about 30-60 minutes on 3-second cycle, if accuracy is still "low", expose the sensor to a solvent vapor (e.g. marker pen, alcohol etc). The AQI should go to about 500 and then the accuracy will probably go to "medium". If not, it needs longer running first.

The accuracy will slowly alternate between medium and high (and occasionally low). This is normal operation ("medium" is actually OK).

If you change settings, exit cycle mode, or reset the board, it will need to recalibrate again.

vincePeKaWay, 2021-07-12:
Alright, I got it working! Thank you very much

closed 2021-07-12 by vincePeKaWay