metriful/sensor #27: Error during comilation of Metriful_sensor.cpp for Arduino and ESP8266

MarcoLeti, 2021-07-15:
:\Program Files (x86)\Arduino\libraries\Metriful_Sensor\Metriful_sensor.cpp:213:31: error: expected ')' before '__INT32'
  213 |     sprintf(strbuf,"%s%u.%u %" PRIu32 " %u.%u %" PRIu32 " ",isPositive?"":"-", T_intPart, T_fractionalPart,
      |            ~                  ^
      |                               )

metriful, 2021-07-16:
Hi,

Thanks for reporting this.

It is an error in the latest version of the ESP8266 library (3.0.0 and later). As a work-around please do the following:

1. In the Arduino IDE menu, go to Tools > Board > Boards Manager. Search for the package named **esp8266 by ESP8266 Community**.

2. In the selection box, choose version **2.7.4** (or earlier) and click "install"

3. The programs will now compile correctly without errors.

NOTE: the IDE will think the library is out of date and may prompt you to update it. Ignore this.

You could report this problem to the developers of the ESP8266 library.

metriful, 2021-07-27:
This problem is now fixed in the new ESP8266 library version 3.0.2

closed 2021-07-27 by metriful