metriful/sensor #34: On Python 3.7 the webserver example code fails due to jinja2 posix path incompatibility

raz0rsedge1975, 2024-01-14:
When I try to run example python code on my Pi Zero2, everything runs fine except the examples simple_webserver and graph_webserver. However, the error message on both scripts is the same:
`Traceback (most recent call last):
  File "graph_web_server.py", line 10, in <module>
    import sensor_package.servers as server
  File "/home/pi/sensor/Python/Raspberry_Pi/sensor_package/servers.py", line 25, in <module>
    class SimpleWebpageHandler(BaseHTTPRequestHandler):
  File "/home/pi/sensor/Python/Raspberry_Pi/sensor_package/servers.py", line 42, in SimpleWebpageHandler
    loader=jinja2.FileSystemLoader(Path(__file__).parent),
  File "/usr/lib/python3/dist-packages/jinja2/loaders.py", line 163, in __init__
    self.searchpath = list(searchpath)
TypeError: 'PosixPath' object is not iterable`

Is there a knows workaround to solve this issue quickly?

Thanks for your awesome work!

raz0r

metriful, 2024-01-15:
Hi. You need to update your jinja2 package. First make a note of the current version in case you need to revert:

1. Do: ```pip freeze | grep Jinja2``` and note the version number displayed (e.g. 1.2.3)
2. Update the package with: ```pip install -U Jinja2```
3. Test the programs to see if they now run OK.

If you get problems, try installing the exact version we use for testing:
```pip install Jinja2==2.11.3```
If the update causes other problems, you can revert to the original version using the same command.

raz0rsedge1975, 2024-01-20:
Interestingly enough, this only happened to a fresh install on a pi2zero - thought everything is up-to-date already.
`pip install -U Jinja2` fixed the problem, thanks for this quick solution.

metriful, 2024-01-22:
Great, thanks for the update.

closed 2024-01-22 by metriful