metriful/sensor #40: Question: ESPHome Realtime Data

DerT94, 2025-01-10:
Hey,
I now have the sensor here and have been able to turn it into Home Assistant and read out data. However, I have a question of understanding. I would like to have near real-time data for light and sound. To do this, the sensor would have to transmit the data to Home Assistant when a threshold value is changed. If I interpret it correctly, the code used for ESP Home does not offer this function.
What would be the best way to achieve this? 
Thank you very much

metriful, 2025-01-10:
Hi, well done for getting it working!

It sounds like you want to trigger data transmission to Home Assistant, depending on sound/light values. This is not really compatible with ESPHome (at least with the way we set it up) because it works on a regular transmission cycle.

You may be able to do it by using the Home Assistant "POST" example (`sensor/Arduino/Examples/Home_Assistant`) instead of ESPHome.
You would need to edit the example to do the "sendNumericData" in response to sound/light values. You could also use the interrupt system to respond more quickly (see the example `sensor/Arduino/Examples/interrupts`). This avoids being limited by the 3/100/300 second sensor readout cycle.

DerT94, 2025-01-14:
Hello,
thanks for the answer.
I am now trying a different approach which is also promising.

I have established a new function. (The code is still in a test phase and therefore messy :))
This activates the LIT pin when the threshold is undershot.
This information is enough for me for now.
```
void configureLightSensorForThreshold(uint8_t dev_addr_7bit) {

    uint16_t Threshold_int = 100;
    uint8_t Threshold_fr_2dp = 0;

    Serial.printf("Threshold values: Integer = %u, Decimal = %u\n", Threshold_int, Threshold_fr_2dp);

    // Schwellenwert setzen
    setLightInterruptThreshold(dev_addr_7bit, Threshold_int, Threshold_fr_2dp);  

    // Interrupt-Typ setzen
    uint8_t interrupt_type = LIGHT_INT_TYPE_COMP;
    TransmitI2C(dev_addr_7bit, LIGHT_INTERRUPT_TYPE_REG, &interrupt_type, 1);

    // Polarität setzen
    uint8_t polarity = LIGHT_INT_POL_POSITIVE;
    TransmitI2C(dev_addr_7bit, LIGHT_INTERRUPT_POLARITY_REG, &polarity, 1);

    // Interrupt aktivieren
    uint8_t interrupt_enable = 1;
    TransmitI2C(dev_addr_7bit, LIGHT_INTERRUPT_ENABLE_REG, &interrupt_enable, 1);
}

```

Now I would like to adjust the threshold value via HomeAssistant.
The function is called in a set_action at a number sensor
Code:
```
void updateThresholds(uint16_t threshold_int, uint8_t threshold_fr_2dp)
{
  // Interrupt activate
  uint8_t interrupt_enable_d = 0;
  TransmitI2C(I2C_ADDRESS, LIGHT_INTERRUPT_ENABLE_REG, &interrupt_enable_d, 1);

  ESP_LOGI("Threshold", "Updating Light Interrupt Threshold with Integer: %d, Decimal: %d", threshold_int, threshold_fr_2dp);
  // Aufruf der bestehenden Funktion mit den übergebenen Werten
  setLightInterruptThreshold(I2C_ADDRESS, threshold_int, threshold_fr_2dp);

  // Interrupt aktivieren
  uint8_t interrupt_enable_e = 1;
  TransmitI2C(I2C_ADDRESS, LIGHT_INTERRUPT_ENABLE_REG, &interrupt_enable_e, 1);
}
```

I plan to have all the settings of the interface set via Home Asisstant.
Any tips?

If it works well I would like to make the code available here.

metriful, 2025-01-15:
OK, keep going and reply if you have a problem!

One tip: don't change many things simultaneously. It's best to start with a working example, then make small changes gradually and re-test after each change.

closed 2025-02-10 by metriful