metriful/sensor #42: Individual Sensor Values

pjshots, 2025-02-06:
Hi, I'd like to extract individual sensor values to send them off to an LCD or LED display. I can't seem to be able to do that, do you have a quick way? I'm using an ESP8266. Thanks.

pjshots, 2025-02-06:
Its OK I found out how to do it. Love the sensor package. Thanks.

metriful, 2025-02-10:
Hi,

Start by making a working program to display text with a numeric variable on your display, without using the sensor. A simple example might be:

```
uint16_t light = 123;
lcd.print("Light = ");
lcd.print(light);
lcd.print(" lux");
```

Also make sure it works in a loop, with a pause and a changing value (e.g. light+=1) on each loop.

Then you can combine this with one of our software examples: "cycle_readout" would be best. Replace the variable with the data read from the sensor.

For example, cycle_readout.ino line 90 has:

`lightData = getLightData(I2C_ADDRESS);`

then after this you can add
```
lcd.print("Light = ");
lcd.print(lightData.illum_lux_int);
lcd.print(" lux");
```

A more advanced way could be to copy/edit the example print functions (like "printLightData") and the functions they call, and replace the `Serial.print` calls with `lcd.print` or similar.

Use separate pins for the sensor and for the display (don't try to share the I2C bus).

closed 2025-02-10 by metriful