metriful/sensor #44: no air quality or illuminance data

amitdmore, 2025-05-05:
Unable to retrieve air quality or illuminance data from the sensor. i am constantly getting N/A values back

Below is the output of my code:

📷 Sending ON_DEMAND_MEASURE_CMD...
⌛ Waiting 2.5 seconds for data to be ready...

✅ Raw air_data dictionary:
{'T_C': 25.0, 'P_Pa': 100959, 'H_pc': 40.0, 'G_ohm': 42613, 'T_F': 77.0, 'F_unit': '°F', 'C_unit': '°C', 'T': 25.0, 'T_unit': '°C'}

📊 Formatted MS430 Sensor Readings:
🌡️  Temperature: 25.0 °C
💧 Humidity:    40.0 %
🧪 CO₂ (eCO₂):  N/A ppm
🔬 TVOC:        N/A ppb
🧠 AQI Accuracy Level: 0 (0=not ready, 3=fully calibrated)
💡 Illuminance: N/A lux
💡 White light: 156
🔁 Waiting 30 seconds before next reading...

📷 Sending ON_DEMAND_MEASURE_CMD...
⌛ Waiting 2.5 seconds for data to be ready...

✅ Raw air_data dictionary:
{'T_C': 25.0, 'P_Pa': 100959, 'H_pc': 40.1, 'G_ohm': 42829, 'T_F': 77.0, 'F_unit': '°F', 'C_unit': '°C', 'T': 25.0, 'T_unit': '°C'}

📊 Formatted MS430 Sensor Readings:
🌡️  Temperature: 25.0 °C
💧 Humidity:    40.1 %
🧪 CO₂ (eCO₂):  N/A ppm
🔬 TVOC:        N/A ppb
🧠 AQI Accuracy Level: 0 (0=not ready, 3=fully calibrated)
💡 Illuminance: N/A lux
💡 White light: 154
🔁 Waiting 30 seconds before next reading...

📷 Sending ON_DEMAND_MEASURE_CMD...
⌛ Waiting 2.5 seconds for data to be ready...

i am running this code for more than week to see if sensor get calibrated but not getting any luck

Below is my python code 

import time
import sys
import smbus

# Add Metriful library path
sys.path.append('/home/amit/mypiclient/sensor/Python/Raspberry_Pi')
import sensor_package.sensor_functions as sensor
import sensor_package.sensor_constants as const

I2C_ADDR = 0x71
bus = smbus.SMBus(1)

# 🔄 Only reset ONCE (important for AQI calibration)
print("🔄 Resetting MS430 sensor...")
bus.write_byte(I2C_ADDR, const.RESET_CMD)
time.sleep(1.5)

# ✅ Start the loop
try:
    while True:
        print("📷 Sending ON_DEMAND_MEASURE_CMD...")
        bus.write_byte(I2C_ADDR, const.ON_DEMAND_MEASURE_CMD)

        print("⌛ Waiting 2.5 seconds for data to be ready...")
        time.sleep(2.5)

        try:
            # 🔹 Read air data
            air_data = sensor.get_air_data(bus)

            print("\n✅ Raw air_data dictionary:")
            print(air_data)

            print("\n📊 Formatted MS430 Sensor Readings:")
            print(f"🌡️  Temperature: {air_data.get('T_C', 'N/A')} {air_data.get('C_unit', '')}")
            print(f"💧 Humidity:    {air_data.get('H_pc', 'N/A')} %")
            print(f"🧪 CO₂ (eCO₂):  {air_data.get('eCO2_ppm', 'N/A')} ppm")
            print(f"🔬 TVOC:        {air_data.get('TVOC_ppb', 'N/A')} ppb")

            # 🔹 AQI calibration accuracy
            try:
                accuracy = bus.read_byte_data(I2C_ADDR, const.AQI_ACCURACY_READ)
                print(f"🧠 AQI Accuracy Level: {accuracy} (0=not ready, 3=fully calibrated)")
            except Exception as e:
                print(f"⚠️ Could not read AQI accuracy: {e}")

            # 🔹 Light data
            light_data = sensor.get_light_data(bus)
            print(f"💡 Illuminance: {light_data.get('lux', 'N/A')} lux")
            print(f"💡 White light: {light_data.get('white', 'N/A')}")

        except Exception as e:
            print(f"❌ Failed to read sensor data: {e}")

        print("🔁 Waiting 30 seconds before next reading...\n")
        time.sleep(3)

except KeyboardInterrupt:
    print("\n🛑 Stopped by user.")

Any help would be much appreciated

metriful, 2025-05-07:
Hi,

There are two separate problems here:

1. The missing illuminance value is because your line containing `light_data.get('lux', 'N/A')` should instead use `light_data.get('illum_lux', 'N/A')`

2. The air quality data (TVOC, eCO2, bVOC, accuracy) are not measured in the on-demand mode, which you are using. You need to use the cycle-mode to get these. This is because the Bosch air quality sensor uses an algorithm that needs to repeat strictly every 3 or 100 or 300 seconds.

To get the air quality measurements, please try using the cycle_readout.py example with 100 second cycle time. Do this initially without any changes to our code, to check you can get it working, then make modifications if you like. 

In cycle mode, the Raspberry Pi waits for a signal from the MS430, then reads the data, then waits for the next signal etc. It's important to use the `GPIO.event_detected(sensor.READY_pin)` for this waiting, because using `time.sleep()` will eventually lose synchronisation between the Pi and MS430.

amitdmore, 2025-05-07:
Thank you, i am now able to get the data out from sensor. last question how long its gone take to calibrate it as i am running for more than 48 hours for every 3 seconds but still it has not calibrated.

metriful, 2025-05-08:
The air quality measurement accuracy can be slow to increase, especially with new sensors, for two reasons:

1. The air quality sensor uses a small internal heater - this will gradually evaporate off impurities when used after a period of storage.

2. The analysis algorithm self-calibrates in response to a range of air qualities, which must be provided to the sensor after it starts monitoring.

To speed up this process:

- Run any code which repeatedly measures the air quality and keep it running as long as possible, ideally at least 48 hours.
- If the accuracy is low (0 or 1) after running for an hour, expose the sensor to polluted air for a few minutes. A solvent vapor such as from a marker pen or alcohol is ideal.

In normal use the accuracy does not remain on 3 (highest) all the time but instead will periodically decrease/increase as calibration is ongoing. Calibration restarts when the sensor is reset.

amitdmore, 2025-05-08:
Thank You for your response

closed 2025-05-08 by amitdmore